tessel/t1-firmware #125: SPI bitOrder value

LinusU, 2015-01-13:
When specifying a `bitValue` of `'msb'` (which is the default) the tessel throws an error at me.

```javascript
this.spi = new port.SPI({
  clockSpeed: 4*1E6,
  cpol: 0,
  cpha: 0,
  bitOrder: 'msb',
  chipSelect: this.CS,
  chipSelectActive: 'low'
});
```

```text
Error: Invalid property value
    at _new ([string "colony-init.lua"]:1)
    at propertySetWithDefault ([string "tessel.js"]:49)
    at f ([string "tessel.js"]:1183)
    at _new ([string "colony-init.lua"]:1)
    at f ([string "tessel.js"]:1425)
    at _new ([string "colony-init.lua"]:1)
    at f (/app//index.js:34)
    at _new ([string "colony-init.lua"]:1)
    at use (/app//index.js:78)
    at res (/app/test.js:5)
```

Removing just `bitOrder: 'msb',` fixes it...

jiahuang, 2015-01-13:
Looks like bitOrder is not specified by string: https://github.com/tessel/firmware/blob/master/builtin/tessel.js#L1183

https://github.com/tessel/firmware/blob/master/builtin/tessel.js#L1336-L1339

Does it work if you try this way?

LinusU, 2015-01-14:
It is, but the string should be uppercase. Using `'MSB'` works.

This could be treated as a documentation issue in which case the documentation should be updated. Or we could make the function accept both upper- and lowercase.

jiahuang, 2015-01-14:
Oh yeah, my bad, didn't look at the propertySetWithDefault function properly.

Yeah, the function should be able to take in both upper and lowercase. Do you want to take this change?

LinusU, 2015-01-14:
Sure, I'll fix it now...

commit 70ff1a72 (2015-01-29): Merge #126 r=@jiahuang: spi: accept bitOrder in lowercase

Fixes #125
PR #126 (2015-01-29): spi: accept bitOrder in lowercase
Fixes #125
closed 2015-01-29 by tm-rampart