tessel/t1-firmware #130: Error sending Keen.io request

davidcaseria, 2015-02-05:
At first I attempted to send data to Keen on my Tessel using the [keen-js](https://github.com/keen/keen-js) module but for some reason my code seemed to be hanging. By hanging I mean it printed true (since I log if it's connected to WiFi) and then never completed the Keen request.

I tested basically the same code through just running Node.js locally on my computer and the code executed fine.

Now, I'm trying to use the older [keen.io](https://github.com/keenlabs/KeenClient-Node) client. However, I'm getting an error when running the code on my Tessel: `Error: ENOENT`.

Running the code on my computer using Node.js works however.

```js
var tessel = require('tessel');
var wifi = require('wifi-cc3000');

var Keen = require('keen.io');
var keen = Keen.configure({
  projectId: '',
  writeKey: ''
});

function send() {
  var results = {
    temperature: 71,
    humidity: 30
  };
  console.log('WiFi: ' + wifi.isConnected());
  keen.addEvent('indoor_climate', results, function (err) {
    if (err) {
      console.log(err);
    } else {
      console.log('added event', results);
    }
    setTimeout(send, 10000);
  })
}
send();
```

johnnyman727, 2015-02-05:
Hi @davidcaseria, could you run this same script with the `--listen=all` flag and post the results here?

davidcaseria, 2015-02-05:
```
TESSEL! Connected to TM-00-04-f000da30-00604f52-541165c2.

INFO Bundling directory /Users/davidcaseria/Desktop/tessel
INFO Deploying bundle (1.07 MB)...

Tessel is reading 1118208 bytes...
Populating filesystem...
Starting script: /_start.js
Initializing runtime...
Loading tessel library...
Running script...
Uptime since startup: 62.127148s
INFO Running script...
WiFi: true
IP Address: 192.168.1.16
DNS: 192.168.1.1
DHCP server: 192.168.1.1
Default Gateway: 192.168.1.1
Connected to WiFi!
Opening socket 0
tSLInformation.usNumberOfFreeBuffers 6
HCI Event error on command: 1008
Closing socket 0
Opening socket 0
tSLInformation.usNumberOfFreeBuffers 5
HCI Event error on command: 1008
HCI Event error on command: 100b
Closing socket 0
Opening socket 0
tSLInformation.usNumberOfFreeBuffers 6
HCI Event error on command: 1008
Closing socket 0
Error: ENOENT
```

johnnyman727, 2015-02-05:
Interesting... I don't see either of those error codes (0x1008 or 0x100b) on the [HCI Error Description Page-(http://processors.wiki.ti.com/index.php/CC3000_HCI_EVNT_messages). @jiahuang any ideas?

closed 2024-05-13 by davidcaseria