tessel/t1-firmware #141: gps_get_time actuall calls gps_get_date

srl295, 2015-07-07:
https://github.com/tessel/t1-firmware/blob/master/src/hw/l_hw.c#L735

Shouldn't this be `gps_get_date()`? Looks like a copy and paste error.
```
> var hw = process.binding('hw'); [ hw.gps_get_time(), hw.gps_get_date() ]
[ 213053.119, 213053.119 ]
```

Related: https://github.com/tessel/gps-a2235h/issues/3

srl295, 2015-07-09:
btw the above fix did seem to work for me, except that somehow the time and datestamps became integer (no fractional seconds).

johnnyman727, 2015-07-17:
@srl295 thanks for catching this! Do you think you'll be able to open a PR and roll that change in?

srl295, 2015-08-08:
@johnnyman727 yes i will open a PR!

closed 2015-08-10 by johnnyman727

yetzt, 2016-02-05:
after merging the fix, when will there be a new firmware to update my tessel?

johnnyman727, 2016-02-09:
@yetzt I will try to put out a new release this week. As you may have noticed, nobody else is actively working on T1 so the process of putting out a new build is a bit of a forgotten art. I also want to change one API that is incompatible with T2 so we may be able to kill two birds with one stone.

yetzt, 2016-02-10:
@johnnyman727 i noticed. thanks for caring! <3

johnnyman727, 2016-02-18:
@yetzt new build should be available now (`0.2.0`)

yetzt, 2016-02-18:
\o/ <3 thanks @johnnyman727

srl295, 2016-02-18:
Ooh- thanks!